keesverruijt/BR24radar_pi #233: Range is km although settings is Nm

Hakansv, 2016-09-26:
Today's fetch. Nice job and good cleaned - neither warnings nor errors. Although I've tested trails_performance I'll make live test tomorrow.

The range unit is km while settings is NM
[image]

douwefokkema, 2016-09-26:
I can't reproduce this issue yet....I will check the code.

Hakansv, 2016-09-26:
Douwe. You couldn't. - Hmmm - I'm at home using emulator, could that be the cause. I'll check on board tomorrow.

canboat, 2016-09-26:
This could be related to difference between 4g and br24. The code needs to take the distance in decimeters sent by the radar and map that back to a range, it uses a double lookup but this does depend on it being able to find the exact value right now. This was done so that if you set O to meters but the plotter to NM or viceversa and then let the non O product set a range that the plugin reflects the range selected on that other product. I think this is not the right decision actually.

Hakansv, 2016-09-27:
This was a simulation, emulator, problem only. Not an issue using a real BR24 radar.

closed 2016-09-27 by Hakansv