keesverruijt/BR24radar_pi #234: Range selection + - button has opposite functions

Hakansv, 2016-09-27:
When clicking - the range is increasing and opposite for the + button.

canboat, 2017-01-15:
Yes, this way it is like all Maps applications and the chartplotters.

Hakansv, 2017-01-15:
I don't fully remember this task but have some idea it was changed at the time and I can't check while on simulation. I'll come back after a boat visit.

canboat, 2017-01-15:
Met vriendelijke groeten,
Kees

Hakansv, 2017-01-20:
I've been in the boat to check this.
Yes, the + sign is still increasing the range and vice verse. I do agree a + sign is logical when talking about a zoom function, like zooming in OCPN. But since the designation or header says "range" equal a distance it's not logical for me.
But if it's normally used the way you've described it's not a big deal for me. The +/- functions are easily detected.
Kees

closed 2017-01-20 by Hakansv