keesverruijt/BR24radar_pi #235: Radar (range) ring is missing

Hakansv, 2016-09-27:
There is a commit about radar ring at extreme ranges. I understand that as the range distance ring - correct?
If so there is no range ring at any range on BR24.

canboat, 2017-01-15:
I think you are referring to how we used to mark the outer range received byte as "full echo" (e.g., "red"). This was useful when we were still verifying that we were receiving proper data from the radome and drawing it at the correct distance.

It's been removed for a while, there are "real" radar range rings (4) on the PPI window.

Do you miss it?

Hakansv, 2017-01-15:
Well, there's a tiny ring just now on the chart overlay and I'm very satisfied with that, thanks,

closed 2017-01-15 by Hakansv