keesverruijt/BR24radar_pi #237: compile error:  error: ‘class wxMenuItem’ has no member named ‘SetFont’; did you mean ‘SetText’?

LeaseOnLife, 2017-01-22:
even after a clean clone, I am getting this:

 50%] Building CXX object CMakeFiles/br24radar_pi.dir/src/br24ControlsDialog.cpp.o
[ 51%] Building CXX object CMakeFiles/br24radar_pi.dir/src/br24MessageBox.cpp.o
[ 53%] Building CXX object CMakeFiles/br24radar_pi.dir/src/br24OptionsDialog.cpp.o
[ 55%] Building CXX object CMakeFiles/br24radar_pi.dir/src/br24Receive.cpp.o
[ 56%] Building CXX object CMakeFiles/br24radar_pi.dir/src/RadarMarpa.cpp.o
/CODING/PLUGINS/BR24radar_pi/src/RadarMarpa.cpp: In member function ‘void br24::RadarArpa::RefreshArpaTargets()’:
/CODING/PLUGINS/BR24radar_pi/src/RadarMarpa.cpp:395:18: error: ‘class wxMenuItem’ has no member named ‘SetFont’; did you mean ‘SetText’?
             mi5->SetFont(*qFont);
                  ^~~~~~~
/CODING/PLUGINS/BR24radar_pi/src/RadarMarpa.cpp:396:18: error: ‘class wxMenuItem’ has no member named ‘SetFont’; did you mean ‘SetText’?
             mi6->SetFont(*qFont);
                  ^~~~~~~
CMakeFiles/br24radar_pi.dir/build.make:254: recipe for target 'CMakeFiles/br24radar_pi.dir/src/RadarMarpa.cpp.o' failed
make[2]: *** [CMakeFiles/br24radar_pi.dir/src/RadarMarpa.cpp.o] Error 1
CMakeFiles/Makefile2:163: recipe for target 'CMakeFiles/br24radar_pi.dir/all' failed
make[1]: *** [CMakeFiles/br24radar_pi.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

cmake shows this:
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- *** Staging to build br24radar_pi ***
-- Found wxWidgets: -L/usr/lib/x86_64-linux-gnu;-pthread;;;-lwx_baseu-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_gl-3.0 (found version "3.0.2") 
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found OpenGL...
--     Lib: /usr/lib/x86_64-linux-gnu/libGLU.so/usr/lib/x86_64-linux-gnu/libGL.so
--     Include: /usr/include
-- Found Gettext: /usr/bin/msgmerge (found version "0.19.8.1") 
-- Found BZip2: /usr/lib/x86_64-linux-gnu/libbz2.so (found version "1.0.6") 
-- Looking for BZ2_bzCompressInit
-- Looking for BZ2_bzCompressInit - found
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Configuring done
-- Generating done
-- Build files have been written to: /CODING/PLUGINS/BR24radar_pi/build

Hakansv, 2017-01-22:
LeaseOnLife
 I've built and run it in both Windows and Ubuntu w/o problem. In Windows I've built it against a opencpn.lib from OCPN v4.5. In fact you also need resent O 4.5 to get a proper run.

SetFont would not be linked to wxMenuItem but to wxOwnerDrawnBase?? Any Linux guru here to can solve this error?
Håkan

Hakansv, 2017-01-22:
Sorry --- I must have built this on my Linux before the menu item was moved. The SetFont would not be there for Linux but only for Windows
Douwe-- I think it would be like this in RadarMarpa.cpp row 394
#ifdef __WXMSW__
wxFont *qFont = OCPNGetFont(_("Menu"), 10);
            mi5->SetFont(*qFont);
            mi6->SetFont(*qFont);
            m_pi->m_context_menu_delete_marpa_target = AddCanvasContextMenuItem(mi5, m_pi);
            m_pi->m_context_menu_delete_all_marpa_targets = AddCanvasContextMenuItem(mi6, m_pi);
#endif
}

LeaseOnLife  could you try this patch?

LeaseOnLife, 2017-01-22:
that compiled! Thanks Håkan

Dirk

Hakansv, 2017-01-23:
Good.
But actually I was to hasty and slipshod. You'll not get any menu items for ARPA targets in OCPN. The patch should be like this:
```
#ifdef __WXMSW__
            wxFont *qFont = OCPNGetFont(_("Menu"), 10);
            mi5->SetFont(*qFont);
            mi6->SetFont(*qFont);
#endif
```
This patch is now pull requested to Canboat master.

LeaseOnLife, 2017-01-23:
checked out latest, compiled, closing issue
Dirk

closed 2017-01-23 by LeaseOnLife