keesverruijt/BR24radar_pi #241: Heading on radar is not applied on a per spoke basis

douwefokkema, 2017-06-04:
Heading on radar is not applied on a per spoke basis to m_hdt (the internal heading) but only once per second (through notify). As the spoke bearings are calculated using m_hdt, the effect of heading on radar is lost, resulting in unstable or not functioning ARPA with a moving ship.

marcus-aa, 2017-06-05:
I have seen this, but I thought it was something wrong in my system... 

If you turn fast  it is very noticeable, to me it almost looks like the whole radar image is rotated "after the fact" but that shouldn't be possible? Could that be what I see once per second?

Every spoke should be "locked" after drawing it in a certain heading ? Because when turning the boat, already drawn spokes must never be changed.

BR /Marcus

douwefokkema, 2017-06-05:
Marcus, you are completely right. This problem was not only for (M)ARPA but for the overlay itself (requiring a heading) as well. The way it should be is that every spoke gets a heading with it from the radar. This heading  should be used to get the bearing to draw that spoke on the chart. This was right in earlier versions, but got mixed up in a recent restructuring of the code. Solution available now  on https://github.com/douwefokkema/BR24radar_pi. Only tested with simulation.

canboat, 2017-06-05:
HI Douwe, can you send me a pull request? 👍

closed 2017-06-05 by douwefokkema