keesverruijt/BR24radar_pi #246: Multi Sweep filter kills the RADAR image

douwefokkema, 2017-07-01:
"Multi Sweep filter" (from the main_menu->View) kills the RADAR image except blue lines around some ARPA targets.

douwefokkema, 2017-07-01:
Should we fix the Multi Sweep filter or just take it out?
Multi Sweep filter is meant to suppress accidental returns by only showing pixels that are seen 2 out of 3 sweeps. Fast moving targets may be lost. 
A better way to avoid accidental alarms in a guard zone is  to use ARPA. ARPA will register a target and pass it to the AIS handling of OCPN. Moving targets will be followed with speed and heading.

Hakansv, 2017-07-01:
For me we can take it out. I've never used it so far. Mostly because I haven't used the radar in as big waves as they are creating trouble for the sea filter. That was once the reason to introduce the Multi Sweep filter if remember correct. 
If I'm in the need of a guard alarm I'll also have the CPA guard activated in OCPN so ARPA targets from  BR24 ARPA zone(s) will be good enough.

closed 2017-08-07 by canboat