keesverruijt/BR24radar_pi #248: Radar plugin crashes at shutdown on Windows 10 Creators edition

douwefokkema, 2017-07-29:
Crash occurs most of the time but not always. Crash on m_receive->Wait(); after  m_receive->Shutdown(); for radar B only. Solved by removing the  m_receive->Shutdown(); 
Solution on https://github.com/douwefokkema/BR24radar_pi/

canboat, 2017-08-07:
I changed the order in which various things happen in https://github.com/canboat/BR24radar_pi/commit/5925586c78b59f8bea277d0cf9c21da57f4186d6 which should fix this. The two Shutdown() calls are now done early, and then we wait for both threads to finish.

douwefokkema, 2017-08-10:
Still crashing on shutdown. Unless wait is removed, then fine.
if (m_receive) {
     wxLongLong threadStartWait = wxGetUTCTimeMillis();
-    m_receive->Wait();
+  //  m_receive->Wait();
     wxLongLong threadEndWait = wxGetUTCTimeMillis();
     wxLongLong threadExtraWait = 0;

canboat, 2017-08-10:
I will check in code that shuffles the shutdown code to stop the threads earlier, before we even close dialogs. This will come as an overall cleanup commit instigated by _valgrind_ and its _helgrind_ race condition detection module.

canboat, 2017-08-10:
Please try again with https://github.com/canboat/BR24radar_pi/commit/be5d0a05269805729c7ef25ff969ebdc73f5e47d

If this doesn't help then we may have to add a config file option to not do the Wait(). I do not feel comfortable leaving it out -- it does actually clean up stuff as well as waiting.

douwefokkema, 2017-08-12:
Perfect shutdown after be5d0a0 !

closed 2017-08-15 by canboat