keesverruijt/BR24radar_pi #252: Build error in Windows VS2013

Hakansv, 2017-08-06:
Hi Kees. Exiting development. Build errors in Windows though:
4>     Creating library C:/Builds/OCPN/BR24radar_pi/build/Release/br24radar_pi.lib and object C:/Builds/OCPN/BR24radar_pi/build/Release/br24radar_pi.exp
4>RadarCanvas.obj : error LNK2019: unresolved external symbol "void __cdecl PlugInAISDrawGL(class wxGLCanvas *,class PlugIn_ViewPort const &)" (?PlugInAISDrawGL@@YAXPAVwxGLCanvas@@ABVPlugIn_ViewPort@@@Z) referenced in function "public: void __thiscall br24::RadarCanvas::Render(class wxPaintEvent &)" (?Render@RadarCanvas@br24@@QAEXAAVwxPaintEvent@@@Z)
4>RadarCanvas.obj : error LNK2019: unresolved external symbol "class wxColour __cdecl PlugInGetFontColor(class wxString)" (?PlugInGetFontColor@@YA?AVwxColour@@VwxString@@@Z) referenced in function "public: void __thiscall br24::RadarCanvas::Render(class wxPaintEvent &)" (?Render@RadarCanvas@br24@@QAEXAAVwxPaintEvent@@@Z)
4>RadarCanvas.obj : error LNK2019: unresolved external symbol "bool __cdecl PlugInSetFontColor(class wxString,class wxColour)" (?PlugInSetFontColor@@YA_NVwxString@@VwxColour@@@Z) referenced in function "public: void __thiscall br24::RadarCanvas::Render(class wxPaintEvent &)" (?Render@RadarCanvas@br24@@QAEXAAVwxPaintEvent@@@Z)
4>C:\Builds\OCPN\BR24radar_pi\build\Release\br24radar_pi.dll : fatal error LNK1120: 3 unresolved externals

canboat, 2017-08-06:
You need to compile / generate / download the latest opencpn.lib since this now requires API 1.14. The documentation on the new OpenCPN site is completely buggered compared to the old one though so I don’t remember how to get this. Maybe search the forum?

canboat, 2017-08-06:
I checked the pages referred to in https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_api:plugin_api_versions#plugin_pi_api_versions_and_opencpn_version_compatibility but they seem to point to old opencpn.lib files.

Hakansv, 2017-08-06:
I've opencpn.lib in the build directory. It's created today from recent OpenCPN Github

canboat, 2017-08-06:
Well, those are exactly the three new functions in the 1.14 API that the plugin now calls, so you're still using an old one somewhere.

canboat, 2017-08-06:
Aww, crap, I forgot to add DECL_EXP to the definition in the three lines in ocpn_plugin.h. If you add those and recompile OCPN it will work. This is a bummer since it means we need to wait for another OCPN release cycle, or come up with a way of dynamically loading it.

canboat, 2017-08-06:
I didn't find that since this is not necessary on Mac, and optional on Linux.

Hakansv, 2017-08-06:
OK, but it's little above my head.
We've two ocpn_plugin.h involved. One in OCPN and one in BR24Radar_pi. Where to add?

Can you give me an example how to? Or maybe change and merge in BR24 and I do the same in OCPN?

canboat, 2017-08-06:
OK there is no workaround other than us delivering a copy of opencpn.exe -- the functions are optimized away by the MSVC compiler unless exported with the appropriate mojo.

canboat, 2017-08-06:
I just pushed a new commit. Copy the ocpn_plugin.h from BR24radar_pi/src to OpenCPN/include, then compile OpenCPN with it, then copy opencpn.lib to your target directory in BR24radar_pi, then build the plugin.

Hakansv, 2017-08-06:
Very good. Works fine.
What about a PR to OCPN Github with the changed ocpn_plugin.h?

canboat, 2017-08-06:
Yes, PR to OpenCPN coming up once I _verify_ it works in Linux as well.

Do you get texts with the AIS targets on the PPI window:

[image]

Hakansv, 2017-08-06:
-OK.
- I'm at home but will test on board first thing tomorrow.
- Nice picture!

canboat, 2017-08-06:
OK the pull request has been created: https://github.com/OpenCPN/OpenCPN/pull/869

Hakansv, 2017-08-07:
Fine, Then build error is fixed.

closed 2017-08-07 by Hakansv