keesverruijt/BR24radar_pi #254: ARPA contour detection can get into infinite loop

canboat, 2017-08-07:
To reproduce push up Gain and/or lower Sea State to ridiculous levels, at some point the ARPA loop detection will start hanging.

douwefokkema, 2017-08-09:
Solution for "ARPA contour detection can get into infinite loop #254".
Problem was caused by ARPA search target that continued to look for targets inside the guard zone even when max number of targets was reached. When a target was found however it was omitted as max was reached. Solved in commit a5aace. 
I also reduced the max number of targets to 100, was 200.

douwefokkema, 2017-08-09:
Above solution is not OK. Works fine in the "old"version, but not in the version with ARPA in radar windows. Working on it.

douwefokkema, 2017-08-12:
Change in pull request #260 corrected situation for large number of targets.
In commit "Abort contour detection if it is too long" large targets are neglected. The issue addressed is real, but the solution incorrect, by neglecting targets, the ARPA target detection may run in an infinite loop. Targets that exceed maximum size should be "shortcut" to the max contour size and presented as such. This way target detection will mot find the same pixels again. Pull request has been done for this.

closed 2017-09-06 by canboat