keesverruijt/BR24radar_pi #255: OCPN right click canvas menu is corrupt.

Hakansv, 2017-08-07:
Some BR24 parts of this menu has been destroyed and one function is doubled.
Set MARPA target is double.
"Radar Control..." and the translated "Radar Inställningar" would be the same?
Delete (M)ARPA target is missing
Delete all (M)ARPA targets is missing.
See picture below:

[image]

canboat, 2017-08-08:
I can get the double menu only if I switch languages (but then I also get crashes) or if I have 2 installations of the dll installed.

The "Delete (M)ARPA" records were only shown once you had any (M)ARPA targets, but then they would stay there even if the count dropped to zero. I have now fixed this -- the menu items are there always, but when the target count is zero the delete items are greyed out and cannot be clicked.

Hakansv, 2017-08-10:
Fix confirmed. 
Thanks

closed 2017-08-10 by Hakansv