keesverruijt/BR24radar_pi #264: Sean suggestion

rgleason, 2017-08-12:
Douwe, Sean mentioned that when he worked on optimizing the plugin (he had access to abother boats radar at the time) that there are some other areas where the plugin can be optmized. He said he explained those areas at the time. Do you know what he is talking about? Did it get done?

canboat, 2017-08-12:
Yes, that was done in the 2.0 time frame. I refactored the code so it is now easy for an OpenGL expert to work on the radar display method. In fact it can be copied easily to a different radar project. Before this was heavily mixed up with other code.

canboat, 2017-08-12:
We also did phase 1 of decoupling the Navico specific code from all of the radar independent code. This really needs to be followed up by a phase 2 where we generate a common code base that can then be used by a specific radar code base. The Navico plugin team is willing to do this but we haven't had much input or feedback from other radar plugin developers.

closed 2017-08-15 by canboat