keesverruijt/BR24radar_pi #266: Ignore potential ARPA targets close to existing AIS/ARPA/MARPA targets

canboat, 2017-08-15:
Do not recognise ARPA target closer than a certain distance from an existing target, especially an AIS target. This is to prevent ARPA picking up targets already seen as AIS targets.

marcus-aa, 2017-08-15:
Even better would be if a real target fusion code could be developed. In ECDIS systems, ARPA positions updates AIS target positions. Especially for fast boats with Class B AIS with 30 second update intervals, this can be really useful since they travel quite far between updates.

/Marcus

Hakansv, 2017-08-15:
I'm not sure if I misunderstand this issue but we're actually trying to not view (M)ARPA targets covered by a nearby AIS target. See RadarMarpa.ccp at around row 910:
`// Check for AIS target at (M)ARPA position
      double posOffset = (double)m_pi->m_settings.AISatARPAoffset;
      // Default 18 >> look 36 meters around + 3% of distance to target
      
double dist2target = (3.0 / 100) * (double)pol.r / (double)RETURNS_PER_LINE * m_ri->m_range_meters;
      posOffset += dist2target;
      if (m_pi->FindAIS_at_arpaPos(m_position.lat, m_position.lon, posOffset)) s = L;
      PassARPAtoOCPN(&pol, s);
    }`

It may possible to streamline this function using the facilities of API 1.14 instead of JSON messages and the updated array table?

canboat, 2017-08-15:
hhmm, I hadn't actually looked at that code. I just logged the issue because I had a number of AIS targets show up with a ARPA target right besides it. I did have a slight heading misalignment -- maybe that caused it. The track and "COG/SOG" of the ARPA target matched perfectly, so the Kalman filter is working beautifully.

closed 2017-08-15 by canboat