keesverruijt/BR24radar_pi #279: Add support for rotating mast

rgleason, 2017-09-07:
I noticed this in the Halo radar features.
Do not know what sensor or nmea sentence is ised here. It would be a nice to have feature.

If this can be done, it can probably be used in Tactics_pi too.

douwefokkema, 2017-09-07:
Do you have any idea how many of our (potential) users have their Navico radar fixed on a rotating mast?

canboat, 2017-09-07:
If you have a properly installed Navico system this should happen automatically because the radar spokes contain a magnetic or true bearing.

You will need a H5000 or better to put mast rotation onto the NMEA2000 bus. If it doesn’t and somebody complains then we have a test installation where we can test this.

My Zeus 3 has a ‘use mast rotation’ setting but since I don’t have a mast rotation sensor nor a H5000 computer I can’t test this.

In other works: we’ll look at this when somebody with the necessary equipment to test it appears.

rgleason, 2017-09-07:
I have no idea how many use a rotating mast, we had one request while Thomas R. was working on Tactics_pi.  We asked what the nmea0183 sentence was for mast rotation and wanted to have a file that we could use to test, but never heard back.  I believe this may become more common, but most Racers may use other software and hardware....

Merrimac those sound like good answers to me. The H5000 is expensive. 
So, I assume there is no nmea0183 sentence for this, and it just goes onto the nmea 2000 bus as a manufacturer PGN or whatever, and the sensor has it's own transmitting wires back to the H5000, or the sensor just connects to directly to the Nmea2000 bus and is used by the H5000.  In any case, that sentence may not exist in nmea0183. (!) (?)

Panbo Tacktic  Mast Rotation http://www.panbo.com/archives/2011/02/mast_rotation_a_multi-manufacturer_solution.html    [Expensive](http://www.defender.com/product.jsp?path=-1%7C344%7C2028690%7C2028835&id=744727)

Some discussion on Expedition website http://expedition.boardhost.com/viewtopic.php?pid=506

Other mast rotation systems: http://www.flyorsail.com/Wind_Instruments.html  see the bottom

Tinley has some documentation.... In accord with Nmea 2.3/3  using $IIXDR,A,x.x,D,mastangle*hh

[NMEA Mast Rotation Converter Installation.pdf](https://github.com/canboat/BR24radar_pi/files/1286123/NMEA.Mast.Rotation.Converter.Installation.pdf)

http://www.lcjcapteurs.com/products/wind-masthead-unit/?lang=en  Ultrasonic Marine Wind Instrument & Rotating Mast Sensor

http://www.lcjcapteurs.com/how-to-install-lcj-capteurs-wind-sensor-on-open-cpn/?lang=en How to Install with OpenCPN

I realize this is probably a side issue.