keesverruijt/BR24radar_pi #281: Crash when OpenCPN restarted after sutting down with RADAR ON

NAHANNIV, 2017-09-08:


douwefokkema, 2017-09-08:
Is this reproducible and consistent each time you start O after shutting down with running radar? Do you have a log file of this happening? Can you make a log with VerboseLog=63 in the ini file.
I can not reproduce this.

Hakansv, 2017-09-08:
Win10: I've shut down with a live radar several times without subsequent crashes.

NAHANNIV, 2017-09-08:
[image]

NAHANNIV, 2017-09-08:
08:02:09: 

________
08:02:09: 2017-09-08
08:02:09:  ------- Starting OpenCPN -------
08:02:09: Version 4.8.0 Build 2017-08-11
08:02:09: wxWidgets version: wxWidgets 3.0.2 Microsoft Windows NT 64 bit wxMSW
08:02:09: MemoryStatus:  mem_total: 3897 mb,  mem_initial: 13 mb
08:02:09: SData_Locn is C:\Program Files (x86)\OpenCPN\
08:02:09: PrivateDataDir is C:\ProgramData\opencpn\
08:02:09: Using existing Config_File: C:\ProgramData\opencpn\opencpn.ini
08:02:09: Switch InlandEcdis mode Off
08:02:09: Setting Viewpoint Lat/Lon 12.0027, -61.7351
08:02:09: Setting Ownship Lat/Lon 12.0028, -61.7348
08:02:09: Styles loading from C:\Program Files (x86)\OpenCPN\uidata\styles.xml
08:02:09: No styles found at: C:\ProgramData\opencpn\
08:02:09: No styles found at: C:\ProgramData\opencpn\.opencpn\
08:02:09: Using SVG Icons
08:02:09: Detected display size (horizontal): 480 mm
08:02:09: Getting DefaultSystemLocale...
08:02:09: Found LanguageInfo for: English
08:02:09: System default Language:  en_GB
08:02:09: Config file language:  en_US
08:02:09: Adjusted App language:  en_US
8:02:09 AM: Opencpn language load for:  en_US
8:02:09 AM: Loading catalog for opencpn core.
8:02:09 AM: Getting DefaultSystemLocale...
8:02:09 AM: Found LanguageInfo for: English
8:02:09 AM: Opencpn language set to:  en_US
8:02:09 AM: Old log will be moved to opencpn.log.log
8:02:09 AM: Creating MyFrame...size(1344, 728)  position(-8, -8)
8:02:09 AM: Creating glChartCanvas
8:02:09 AM: Metrics:  m_display_size_mm: 480     wxDisplaySize:  1920:1080   
8:02:10 AM: OpenGL-> Renderer String: Intel(R) HD Graphics
8:02:10 AM: OpenGL-> Version reported:  4.4.0 - Build 20.19.15.4474
8:02:10 AM: OpenGL-> Detected Intel renderer, disabling stencil buffer
8:02:10 AM: OpenGL-> Texture rectangle format: de1
8:02:10 AM: OpenGL-> glGenerateMipmap unavailable
8:02:10 AM: OpenGL-> Detected Windows Intel renderer, disabling Vertexbuffer Objects
8:02:10 AM: OpenGL-> Vertexbuffer Objects unavailable
8:02:10 AM: OpenGL-> Using Framebuffer Objects
8:02:10 AM: OpenGL-> Using FBO Stencil buffer
8:02:10 AM: OpenGL-> Using Depth buffer clipping
8:02:10 AM: OpenGL-> Using s3tc dxt1 compression
8:02:10 AM: OpenGL-> Compressed tile size: 128kb (8:1)
8:02:10 AM: OpenGL-> Minimum cartographic line width:  1.0
8:02:10 AM: OpenGL-> Minimum symbol line width:  1.0
8:02:10 AM: ChartDB Cache policy:  Application target is 1024 MBytes
8:02:10 AM: Loading chart db version: V018
8:02:10 AM: Chartdb: Chart directory list follows
8:02:10 AM:   Chart directory #0: C:\Users\Nahanni\Desktop\Johns\Charts\Cm93 jan 2011
8:02:10 AM:   Chart directory #1: C:\Users\Nahanni\Desktop\Johns\Charts\BermudaBahamas
8:02:10 AM:   Chart directory #2: C:\Users\Nahanni\Desktop\Johns\Charts\EasternCaribean
8:02:10 AM:   Chart directory #3: C:\Users\Nahanni\Desktop\Johns\Charts\ErieOntStLawrenceNovaScotia
8:02:10 AM: GPS Watchdog Timeout is: 6 sec.
8:02:10 AM: Initializing Chart C:\Users\Nahanni\Desktop\Johns\Charts\EasternCaribean\BSBCHART\NC46A_1.KAP
8:02:10 AM: OpenCPN Initialized in 694 ms.
8:02:10 AM: Warning: Color not found CHDRD
8:02:11 AM: DPMM: 4   ScaleFactorExp: 1
8:02:11 AM: Looking for UserIcons at C:\ProgramData\opencpn\UserIcons
8:02:11 AM: Loading navobjects from navobj.xml
8:02:11 AM: Done loading navobjects
8:02:11 AM: Opening NMEA Datastream Serial:COM3
8:02:12 AM: PlugInManager searching for PlugIns in location C:\Program Files (x86)\OpenCPN\plugins
8:02:12 AM: PlugInManager: Loading PlugIn: C:\Program Files (x86)\OpenCPN\plugins\br24radar_pi.dll
8:02:12 AM:   C:\Program Files (x86)\OpenCPN\plugins\br24radar_pi.dll
              API Version detected: 114
              PlugIn Version detected: 1207
8:02:12 AM: BR24radar_pi:  set Target trails = 4
8:02:12 AM: BR24radar_pi: Target trail value 4 = 0 revolutions
[... 272 lines cut ...]
8:02:14 AM: BR24radar_pi: SetNMEASentence $AIVLW,4877.78,N,4877.78,N,,N,,N*45

8:02:14 AM: BR24radar_pi: SetNMEASentence $APHDM,154.5,M*36

8:02:14 AM: BR24radar_pi: SetNMEASentence $GPRMC,120218.00,A,1200.16877,N,06144.09013,W,0.214,236.82,080917,14.81,W,A*0C

8:02:14 AM: BR24radar_pi: SetNMEASentence $APHDM,154.5,M*36

8:02:14 AM: BR24radar_pi: SetNMEASentence $AIHDG,154.0,,,14.0,W*0D

8:02:14 AM: BR24radar_pi: RenderGLOverlay context=00000000
8:02:15 AM: BR24radar_pi: WMM plugin provides new magnetic variation -14.806509
8:02:15 AM: BR24radar_pi: SetPositionFixEx var=-14.000000 var_wd=1
8:02:15 AM: BR24radar_pi: GPS position is now known
8:02:15 AM: BR24radar_pi: SetNMEASentence $APHDM,154.4,M*37

8:02:15 AM: BR24radar_pi: SetNMEASentence $APHDM,154.4,M*37

8:02:15 AM: BR24radar_pi: SetNMEASentence $AIMWV,1.6,R,12.9,N,A*08

8:02:15 AM: BR24radar_pi: RenderGLOverlay context=00000000
8:02:15 AM: BR24radar_pi: Automatic range changed from 0 to 423 meters
8:02:15 AM: BR24radar_pi: Radar A transmit: range 231 meters
8:02:15 AM: Error: BR24radar_pi: Unable to transmit command to unknown radar
8:02:15 AM: BR24radar_pi: RenderRadarOverlay lat=12.0027 lon=-61.7351 v_scale_ppm=2.22679 vp_rotation=0 skew=0 scale=1754.588013 rot=0

NAHANNIV, 2017-09-08:
This is consistently repeatable.
Win 10 Home 1607

canboat, 2017-09-08:
I think I know what the issue is.

We start processing stuff before the configuration file is loaded, and a structure is not initialised before it is used.

I will have a fix for you to try soon.

canboat, 2017-09-08:
JM,

Can you try this download and see if this fixes your crash issue:

https://www.dropbox.com/s/bhqao5276qfpxen/br24radar_pi-3.908-win32.exe?dl=0 <https://www.dropbox.com/s/bhqao5276qfpxen/br24radar_pi-3.908-win32.exe?dl=0>

canboat, 2017-09-08:
Found another issue, please download from the same link above.

NAHANNIV, 2017-09-08:
That works better. Can restart OK RADAR starts, but when I zoom in and then out it crashes.

NAHANNIV, 2017-09-08:
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=2988 hdg=-32768 24 bytes:  18 02 D3 0A 00 44 80 00 AC 0B 00 80 0C 03 5B BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=2990 hdg=-32768 24 bytes:  18 02 D4 0A 00 44 80 00 AE 0B 00 80 0C 03 5D BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=2992 hdg=-32768 24 bytes:  18 02 D5 0A 00 44 80 00 B0 0B 00 80 0C 03 5E BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=2994 hdg=-32768 24 bytes:  18 02 D6 0A 00 44 80 00 B2 0B 00 80 0C 03 5F BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=2996 hdg=-32768 24 bytes:  18 02 D7 0A 00 44 80 00 B4 0B 00 80 0C 03 61 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=2998 hdg=-32768 24 bytes:  18 02 D8 0A 00 44 80 00 B6 0B 00 80 0C 03 62 BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3000 hdg=-32768 24 bytes:  18 02 D9 0A 00 44 80 00 B8 0B 00 80 0C 03 63 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3002 hdg=-32768 24 bytes:  18 02 DA 0A 00 44 80 00 BA 0B 00 80 0C 03 64 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3004 hdg=-32768 24 bytes:  18 02 DB 0A 00 44 80 00 BC 0B 00 80 0C 03 66 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3006 hdg=-32768 24 bytes:  18 02 DC 0A 00 44 80 00 BE 0B 00 80 0C 03 67 BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3008 hdg=-32768 24 bytes:  18 02 DD 0A 00 44 80 00 C0 0B 00 80 0C 03 68 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3010 hdg=-32768 24 bytes:  18 02 DE 0A 00 44 80 00 C2 0B 00 80 0C 03 69 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3012 hdg=-32768 24 bytes:  18 02 DF 0A 00 44 80 00 C4 0B 00 80 0C 03 6B BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3014 hdg=-32768 24 bytes:  18 02 E0 0A 00 44 80 00 C6 0B 00 80 0C 03 6B BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3016 hdg=-32768 24 bytes:  18 02 E1 0A 00 44 80 00 C8 0B 00 80 0C 03 6D BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3018 hdg=-32768 24 bytes:  18 02 E2 0A 00 44 80 00 CA 0B 00 80 0C 03 6F BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3020 hdg=-32768 24 bytes:  18 02 E3 0A 00 44 80 00 CC 0B 00 80 0C 03 70 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3022 hdg=-32768 24 bytes:  18 02 E4 0A 00 44 80 00 CE 0B 00 80 0C 03 70 BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3024 hdg=-32768 24 bytes:  18 02 E5 0A 00 44 80 00 D0 0B 00 80 0C 03 72 BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3026 hdg=-32768 24 bytes:  18 02 E6 0A 00 44 80 00 D2 0B 00 80 0C 03 73 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3028 hdg=-32768 24 bytes:  18 02 E7 0A 00 44 80 00 D4 0B 00 80 0C 03 74 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3030 hdg=-32768 24 bytes:  18 02 E8 0A 00 44 80 00 D6 0B 00 80 0C 03 75 BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3032 hdg=-32768 24 bytes:  18 02 E9 0A 00 44 80 00 D8 0B 00 80 0C 03 77 BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3034 hdg=-32768 24 bytes:  18 02 EA 0A 00 44 80 00 DA 0B 00 80 0C 03 78 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3036 hdg=-32768 24 bytes:  18 02 EB 0A 00 44 80 00 DC 0B 00 80 0C 03 79 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3038 hdg=-32768 24 bytes:  18 02 EC 0A 00 44 80 00 DE 0B 00 80 0C 03 7A BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3040 hdg=-32768 24 bytes:  18 02 ED 0A 00 44 80 00 E0 0B 00 80 0C 03 7C BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3042 hdg=-32768 24 bytes:  18 02 EE 0A 00 44 80 00 E2 0B 00 80 0C 03 7D BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3044 hdg=-32768 24 bytes:  18 02 EF 0A 00 44 80 00 E4 0B 00 80 0C 03 7E BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3046 hdg=-32768 24 bytes:  18 02 F0 0A 00 44 80 00 E6 0B 00 80 0C 03 7F BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3048 hdg=-32768 24 bytes:  18 02 F1 0A 00 44 80 00 E8 0B 00 80 0C 03 80 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3050 hdg=-32768 24 bytes:  18 02 F2 0A 00 44 80 00 EA 0B 00 80 0C 03 81 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3052 hdg=-32768 24 bytes:  18 02 F3 0A 00 44 80 00 EC 0B 00 80 0C 03 82 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3054 hdg=-32768 24 bytes:  18 02 F4 0A 00 44 80 00 EE 0B 00 80 0C 03 83 BC 00 00 00 80 00 00 00 A0
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3056 hdg=-32768 24 bytes:  18 02 F5 0A 00 44 80 00 F0 0B 00 80 0C 03 85 BC 00 00 00 80 00 00 00 80
11:39:39 AM: BR24radar_pi: Radar A range=780, angle=3058 hdg=-32768 24 bytes:  18 02 F6 0A 00 44 80 00 F2 0B 00 80 0C 03 86 BC 00 00 00 80 00 00 00 80
[... cut ...]
11:39:40 AM: BR24radar_pi: SetNMEASentence $GPRMC,153940.00,A,1200.16338,N,06144.09522,W,0.117,250.18,080917,14.81,W,D*0F

11:39:40 AM: BR24radar_pi: main timer
11:39:40 AM: BR24radar_pi: Radar A: CheckGuardZoneBogeys found=0 confirmed=0
11:39:40 AM: BR24radar_pi: Radar A packets 2/0 spokes 64/0/0 Radar B packets 0/0 spokes 0/0/0 
11:39:40 AM: BR24radar_pi: messagebox decision: show=1 overlay=0 auto_hide=1 opengl=1 radarOn=1 navOn=1
11:39:40 AM: BR24radar_pi: messagebox overlay needs met: HIDE
11:39:40 AM: BR24radar_pi: no change

NAHANNIV, 2017-09-08:
Why do others not have this problem ?

douwefokkema, 2017-09-08:
I am really puzzled. Please try also with VerboseLog= 0, there is a chance that all the logging itself influences results.

NAHANNIV, 2017-09-08:
Tried with VeboseLog=0

It Immediate crashes on restarting after shutting down with RADAR on.

canboat, 2017-09-08:
I think it is related to the new way of calculating trails, which happens even if you have them disabled, so that they show immediately if you change the trails timeout or display method. 

I fixed two issues in that area but not enough. 

Can you try restarting with a really long range selected, or with gain = 0 so that there are no targets, and see if that makes any difference?

NAHANNIV, 2017-09-08:
Tried shutting down with range 36NM and gain 0 a few times.

Seems to take longer before crashing when restarted.

It usually seems to crash when zooming in and out.

NAHANNIV, 2017-09-08:
Here is my OpenCPN.ini file after shutting down with the radar on:

[Settings]
ConfigVersionString=Version 4.8.0 Build 2017-08-11
NavMessageShown=1
UIexpert=1
UIStyle=Traditional
ChartNotRenderScaleFactor=2
ShowStatusBar=1
ShowMenuBar=0
DefaultFontSize=0
Fullscreen=0
ShowCompassWindow=1
SetSystemTime=0
ShowGrid=0
PlayShipsBells=0
SoundDeviceIndex=0
FullscreenToolbar=1
TransparentToolbar=1
PermanentMOBIcon=0
ShowLayers=1
ShowDepthUnits=1
AutoAnchorDrop=0
ShowChartOutlines=0
ShowActiveRouteTotal=1
ShowActiveRouteHighway=1
SDMMFormat=0
DistanceFormat=0
SpeedFormat=0
MostRecentGPSUploadConnection=
ShowChartBar=1
GUIScaleFactor=0
ChartObjectScaleFactor=0
FilterNMEA_Avg=0
FilterNMEA_Sec=1
ShowMag=0
UserMagVariation=0.00
CM93DetailFactor=0
CM93DetailZoomPosX=200
CM93DetailZoomPosY=200
ShowCM93DetailSlider=0
SkewToNorthUp=0
OpenGL=1
SoftwareGL=0
ShowFPS=1
ZoomDetailFactor=0
FogOnOverzoom=0
OverzoomVectorScale=1
OverzoomEmphasisBase=10
UseAcceleratedPanning=1
GPUTextureCompression=1
GPUTextureCompressionCaching=1
GPUTextureDimension=512
GPUTextureMemSize=128
SmoothPanZoom=1
UseRasterCharts=1
UseVectorCharts=0
UseCM93Charts=0
CourseUpMode=0
LookAheadMode=1
COGUPAvgSeconds=15
UseMagAPB=0
OwnshipCOGPredictorMinutes=5
OwnshipCOGPredictorWidth=3
OwnshipHDTPredictorMiles=1
OwnShipIconType=1
OwnShipLength=12
OwnShipWidth=4
OwnShipGPSOffsetX=-1
OwnShipGPSOffsetY=9
[... 758 lines cut ...]
Radar1Zone0AlarmOn=0
Radar1Zone0ArpaOn=0
Radar1Zone1StartBearing=0
Radar1Zone1EndBearing=0
Radar1Zone1OuterRange=0
Radar1Zone1InnerRange=0
Radar1Zone1Type=0
Radar1Zone1AlarmOn=0
Radar1Zone1ArpaOn=0
[Directories]
S57DataLocation=
InitChartDir=C:\\Users\\Nahanni\\Desktop\\Johns\\Charts
GPXIODir=C:\\Users\\Nahanni\\Desktop
TCDataDir=
WMMDataLocation=C:\\Program Files (x86)\\OpenCPN\\plugins\\wmm_pi\\data\\
GRIBDirectory=C:\\Users\\Nahanni\\Desktop
[ChartDirectories]
ChartDir1=C:\\Users\\Nahanni\\Desktop\\Johns\\Charts\\Cm93 jan 2011^
ChartDir2=C:\\Users\\Nahanni\\Desktop\\Johns\\Charts\\BermudaBahamas^171520263602
ChartDir3=C:\\Users\\Nahanni\\Desktop\\Johns\\Charts\\EasternCaribean^108576690584
ChartDir4=C:\\Users\\Nahanni\\Desktop\\Johns\\Charts\\ErieOntStLawrenceNovaScotia^203111750199
[AUI]
AUIPerspective=layout2|name=ChartCanvas;caption=;state=768;dir=5;layer=0;row=0;pos=0;prop=100000;bestw=1920;besth=997;minw=-1;minh=-1;maxw=-1;maxh=-1;floatx=-1;floaty=-1;floatw=-1;floath=-1|name=BR24radar_pi-0;caption=Radar A;state=2099151;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=960;besth=540;minw=256;minh=256;maxw=-1;maxh=-1;floatx=-506;floaty=163;floatw=512;floath=512|name=BR24radar_pi-1;caption=Radar B;state=2099151;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=960;besth=540;minw=256;minh=256;maxw=-1;maxh=-1;floatx=1816;floaty=-3;floatw=512;floath=512|name=7b2effff-a543-41eb-b18f-fe59489cffff;caption=Dashboard;state=2098124;dir=2;layer=1;row=0;pos=0;prop=134660;bestw=150;besth=842;minw=150;minh=842;maxw=-1;maxh=-1;floatx=1836;floaty=512;floatw=150;floath=842|dock_size(5,0,0)=398|dock_size(2,1,0)=260|
[TideCurrentDataSources]
tcds0=C:\\Program Files (x86)\\OpenCPN\\tcdata\\HARMONIC.IDX

NAHANNIV, 2017-09-08:
Is anyone else able to reproduce this problem ?

canboat, 2017-09-08:
Ok the zooming in/out is a good hint.  Watching 'true lies' with family now, a new version should be there for you tomorrow morning.

Hakansv, 2017-09-09:
I'm on Win10 and as said no similar experiences but havn't stressed extensively zooming lately. I'll take present code for a live test on board now and revert.

Hakansv, 2017-09-09:
Been testing also the latest code on my BR24 using Win10 and O 4.8. 
I can still close O with an active radar and restart wo any issues. The radar is directly transmitting after restart.
Zoom in/out is also fine wo issues although my BR24 is in AUTO limited to 24 NM range.

One note though; In radarwindow, after "North up" is "Course up" still viewing north up but next level "Course up (Stabilized)" shows heading up. I use HDG from external compass via O NMEA and WMM.

canboat, 2017-09-09:
Douwe reports that he has a crash, but that is not reproducible by me.

I did fix a number of 'trails' issues in zooming in. These are fixed in the following:

https://www.dropbox.com/s/0pc7ablmojqw44x/br24radar_pi-3.909-win32.exe?dl=0

Can you try again?

If this still crashes, please make a copy of your ini file and remove the entire [AUI] section, this contains the onscreen location for the AUI manager that controls the different panes. We saw that you had Radar A at a negative offset. Do you use multiple screens?

canboat, 2017-09-09:
I shortened this issue by putting the logs and ini files in zip files and attaching those to the comment.

NAHANNIV, 2017-09-09:
The latest version will not load.

Shows Incompatible in the plugin configuration.

canboat, 2017-09-09:
Oops, that was a debug build. Replaced it with a release build, same link.

NAHANNIV, 2017-09-09:
Latest version crashed immediately on restart.

then I deleted [AUI] section from .ini file.

Restarted OK, but crashed when scrolling out.

NAHANNIV, 2017-09-09:
The automatically re-created [AUI] section
[AUI]
AUIPerspective=layout2|name=ChartCanvas;caption=;state=768;dir=5;layer=0;row=0;pos=0;prop=100000;bestw=1920;besth=997;minw=-1;minh=-1;maxw=-1;maxh=-1;floatx=-1;floaty=-1;floatw=-1;floath=-1|name=BR24radar_pi-0;caption=Radar A;state=2099151;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=960;besth=540;minw=256;minh=256;maxw=-1;maxh=-1;floatx=-506;floaty=163;floatw=512;floath=512|name=BR24radar_pi-1;caption=Radar B;state=2099151;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=960;besth=540;minw=256;minh=256;maxw=-1;maxh=-1;floatx=1816;floaty=-3;floatw=512;floath=512|name=7b2effff-a543-41eb-b18f-fe59489cffff;caption=Dashboard;state=2098127;dir=4;layer=0;row=0;pos=0;prop=100000;bestw=150;besth=842;minw=150;minh=842;maxw=-1;maxh=-1;floatx=100;floaty=100;floatw=166;floath=881|dock_size(5,0,0)=398|

douwefokkema, 2017-09-10:
I can now fully reproduce your crash (that is: start up with running radar) using your ini file. With the new version of Kees it is no longer crashing always but sometimes. I did another mod that solves the crash with me completely. But that does not mean it solves the problem on your system. Test version at https://www.dropbox.com/s/skwxn7cmrbvz114/br24radar_pi-3.908-win32-test.exe?dl=0 . Dll only at: https://www.dropbox.com/s/wb50m26ock7n5em/br24radar_pi.dll?dl=0 But when you wait a bit, Kees might have a better implementation.

NAHANNIV, 2017-09-10:
Douwe, I downloaded the .exe version, but it did not fix the issue on my system.
Tried it twice and it crashed almost immediately (without any action from me).

What was the problem you fixed ?
I have re-enabled and located my dashboard display after deleting the [AUI] section of my .ini file, but everything else is the same.

douwefokkema, 2017-09-10:
I did put a wait in the processing of the received data from the radar until all transmit sockets were ready. It seemed that the system when started with running radar started processing before all inits were done. Earlier this produced error messages.
Problem now is that I can't reproduce your errors any longer (after my update). As I could only reproduce with your opencpn.ini, there seems to be a dependency on the ini file. Could you try to throw away (rename) yr opencpn.ini and let OpenCPN make a new one. You will have to set comms and chart settings again. Please let me have again the opencpn.log made with VerboseLog=1.

NAHANNIV, 2017-09-10:
VerboseLog=1
Original .ini

08:01:36: 

________
08:01:36: 2017-09-10
08:01:36:  ------- Starting OpenCPN -------
08:01:36: Version 4.8.0 Build 2017-08-11
08:01:36: wxWidgets version: wxWidgets 3.0.2 Microsoft Windows NT 64 bit wxMSW
08:01:36: MemoryStatus:  mem_total: 3897 mb,  mem_initial: 13 mb
08:01:36: SData_Locn is C:\Program Files (x86)\OpenCPN\
08:01:36: PrivateDataDir is C:\ProgramData\opencpn\
08:01:36: Using existing Config_File: C:\ProgramData\opencpn\opencpn.ini
08:01:36: Switch InlandEcdis mode Off
08:01:36: Setting Viewpoint Lat/Lon 12.0025, -61.7345
08:01:36: Setting Ownship Lat/Lon 12.0027, -61.7349
08:01:36: Styles loading from C:\Program Files (x86)\OpenCPN\uidata\styles.xml
08:01:36: No styles found at: C:\ProgramData\opencpn\
08:01:36: No styles found at: C:\ProgramData\opencpn\.opencpn\
08:01:36: Using SVG Icons
08:01:36: Detected display size (horizontal): 480 mm
08:01:36: Getting DefaultSystemLocale...
08:01:36: Found LanguageInfo for: English
08:01:36: System default Language:  en_GB
08:01:36: Config file language:  en_US
08:01:36: Adjusted App language:  en_US
8:01:36 AM: Opencpn language load for:  en_US
8:01:36 AM: Loading catalog for opencpn core.
8:01:36 AM: Getting DefaultSystemLocale...
8:01:36 AM: Found LanguageInfo for: English
8:01:36 AM: Opencpn language set to:  en_US
8:01:36 AM: Creating MyFrame...size(1344, 728)  position(-8, -8)
8:01:36 AM: Creating glChartCanvas
8:01:36 AM: Metrics:  m_display_size_mm: 480     wxDisplaySize:  1920:1080   
8:01:36 AM: OpenGL-> Renderer String: Intel(R) HD Graphics
8:01:36 AM: OpenGL-> Version reported:  4.4.0 - Build 20.19.15.4474
8:01:36 AM: OpenGL-> Detected Intel renderer, disabling stencil buffer
8:01:36 AM: OpenGL-> Texture rectangle format: de1
8:01:36 AM: OpenGL-> glGenerateMipmap unavailable
8:01:36 AM: OpenGL-> Detected Windows Intel renderer, disabling Vertexbuffer Objects
8:01:36 AM: OpenGL-> Vertexbuffer Objects unavailable
8:01:36 AM: OpenGL-> Using Framebuffer Objects
8:01:36 AM: OpenGL-> Using FBO Stencil buffer
8:01:36 AM: OpenGL-> Using Depth buffer clipping
8:01:36 AM: OpenGL-> Using s3tc dxt1 compression
8:01:36 AM: OpenGL-> Compressed tile size: 128kb (8:1)
8:01:36 AM: OpenGL-> Minimum cartographic line width:  1.0
8:01:36 AM: OpenGL-> Minimum symbol line width:  1.0
8:01:36 AM: ChartDB Cache policy:  Application target is 1024 MBytes
8:01:36 AM: Loading chart db version: V018
8:01:36 AM: Chartdb: Chart directory list follows
8:01:36 AM:   Chart directory #0: C:\Users\Nahanni\Desktop\Johns\Charts\Cm93 jan 2011
8:01:36 AM:   Chart directory #1: C:\Users\Nahanni\Desktop\Johns\Charts\BermudaBahamas
8:01:36 AM:   Chart directory #2: C:\Users\Nahanni\Desktop\Johns\Charts\EasternCaribean
8:01:36 AM:   Chart directory #3: C:\Users\Nahanni\Desktop\Johns\Charts\ErieOntStLawrenceNovaScotia
8:01:36 AM: GPS Watchdog Timeout is: 6 sec.
8:01:36 AM: ChartSymbols loaded from C:\Program Files (x86)\OpenCPN\s57data\chartsymbols.xml
8:01:36 AM: Using s57data in C:\Program Files (x86)\OpenCPN\s57data
8:01:37 AM: Initializing Chart C:\Users\Nahanni\Desktop\Johns\Charts\Cm93 jan 2011\
8:01:37 AM: CM93Composite Chart Root is C:\Users\Nahanni\Desktop\Johns\Charts\Cm93 jan 2011\\
8:01:37 AM: Loaded CM93 Dictionary from C:\Users\Nahanni\Desktop\Johns\Charts\Cm93 jan 2011\\
8:01:37 AM: Loading CM93 cell C:\Users\Nahanni\Desktop\Johns\Charts\Cm93 jan 2011\\02700840/F/03060894.F
8:01:37 AM: OpenCPN Initialized in 1395 ms.
8:01:37 AM: Warning: Color not found CHDRD
8:01:38 AM: DPMM: 4   ScaleFactorExp: 1
8:01:38 AM: Looking for UserIcons at C:\ProgramData\opencpn\UserIcons
8:01:38 AM: Loading navobjects from navobj.xml
8:01:39 AM: Done loading navobjects
8:01:39 AM: Applying NavObjChanges
[... 1181 lines cut ...]
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: main timer
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: main timer
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: main timer
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: Radar A received range request for 231 meters from 169.254.253.199
8:02:22 AM: Loading CM93 cell C:\Users\Nahanni\Desktop\Johns\Charts\Cm93 jan 2011\\02700840/F/03050894.F
8:02:22 AM: BR24radar_pi: Automatic range changed from 421 to 661 meters
8:02:22 AM: BR24radar_pi: Radar A transmit: range 463 meters
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: SetPositionFixEx var=-14.000000 var_wd=1
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: main timer
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: main timer
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: main timer
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: Radar A received range request for 463 meters from 169.254.253.199
8:02:22 AM: BR24radar_pi: Automatic range changed from 661 to 843 meters
8:02:22 AM: BR24radar_pi: Radar A transmit: range 463 meters
8:02:22 AM: BR24radar_pi: Radar A render ARPA targets on overlay with rot=0.000000
8:02:22 AM: BR24radar_pi: main timer

NAHANNIV, 2017-09-10:
OK, I deleted my OpenCPN.ini file.
enabled OpenGL
added connection for NMEA data
Enabled radar plugin.

turned on RADAR
closed OpenCPN
Re Opened OpenCPN

Crash !

NAHANNIV, 2017-09-10:
VerboseLog=1
I had to zoom in and out before it crashed;
Sometimes it crashes without any user input, other times it crashes when I zoom in/out

08:13:07: 

________
08:13:07: 2017-09-10
08:13:07:  ------- Starting OpenCPN -------
08:13:07: Version 4.8.0 Build 2017-08-11
08:13:07: wxWidgets version: wxWidgets 3.0.2 Microsoft Windows NT 64 bit wxMSW
08:13:07: MemoryStatus:  mem_total: 3897 mb,  mem_initial: 13 mb
08:13:07: SData_Locn is C:\Program Files (x86)\OpenCPN\
08:13:07: PrivateDataDir is C:\ProgramData\opencpn\
08:13:07: Using existing Config_File: C:\ProgramData\opencpn\opencpn.ini
08:13:07: Switch InlandEcdis mode Off
08:13:07: Setting Viewpoint Lat/Lon 12.0027, -61.7349
08:13:07: Setting Ownship Lat/Lon 12.0027, -61.7349
08:13:07: Styles loading from C:\Program Files (x86)\OpenCPN\uidata\styles.xml
08:13:07: No styles found at: C:\ProgramData\opencpn\
08:13:07: No styles found at: C:\ProgramData\opencpn\.opencpn\
08:13:07: Using SVG Icons
08:13:07: Detected display size (horizontal): 480 mm
08:13:07: Getting DefaultSystemLocale...
08:13:07: Found LanguageInfo for: English
08:13:07: System default Language:  en_GB
08:13:07: Config file language:  en_US
08:13:07: Adjusted App language:  en_US
8:13:07 AM: Opencpn language load for:  en_US
8:13:07 AM: Loading catalog for opencpn core.
8:13:07 AM: Getting DefaultSystemLocale...
8:13:07 AM: Found LanguageInfo for: English
8:13:07 AM: Opencpn language set to:  en_US
8:13:07 AM: Creating MyFrame...size(1344, 728)  position(-8, -8)
8:13:07 AM: Creating glChartCanvas
8:13:07 AM: Metrics:  m_display_size_mm: 480     wxDisplaySize:  1920:1080   
8:13:08 AM: OpenGL-> Renderer String: Intel(R) HD Graphics
8:13:08 AM: OpenGL-> Version reported:  4.4.0 - Build 20.19.15.4474
8:13:08 AM: OpenGL-> Detected Intel renderer, disabling stencil buffer
8:13:08 AM: OpenGL-> Texture rectangle format: de1
8:13:08 AM: OpenGL-> glGenerateMipmap unavailable
8:13:08 AM: OpenGL-> Detected Windows Intel renderer, disabling Vertexbuffer Objects
8:13:08 AM: OpenGL-> Vertexbuffer Objects unavailable
8:13:08 AM: OpenGL-> Using Framebuffer Objects
8:13:08 AM: OpenGL-> Using FBO Stencil buffer
8:13:08 AM: OpenGL-> Using Depth buffer clipping
8:13:08 AM: OpenGL-> Not Using compression
8:13:08 AM: OpenGL-> Minimum cartographic line width:  1.0
8:13:08 AM: OpenGL-> Minimum symbol line width:  1.0
8:13:08 AM: ChartDB Cache policy:  Application target is 1024 MBytes
8:13:08 AM: Chartlist file not found, config chart dir array is empty.  Chartlist target file is:C:\ProgramData\opencpn\CHRTLIST.DAT
8:13:08 AM: GPS Watchdog Timeout is: 6 sec.
8:13:08 AM: OpenCPN Initialized in 599 ms.
8:13:08 AM: Loading World Chart Q=0 in 0 ms.
8:13:08 AM: Background world map loaded from GSHHS datafiles found in: C:\Program Files (x86)\OpenCPN\gshhs\
8:13:08 AM: Warning: Color not found CHDRD
8:13:09 AM: DPMM: 4   ScaleFactorExp: 1
8:13:09 AM: Looking for UserIcons at C:\ProgramData\opencpn\UserIcons
8:13:09 AM: Loading navobjects from navobj.xml
8:13:09 AM: Done loading navobjects
8:13:10 AM: Opening NMEA Datastream Serial:COM3
8:13:10 AM: PlugInManager searching for PlugIns in location C:\Program Files (x86)\OpenCPN\plugins
8:13:10 AM: PlugInManager: Loading PlugIn: C:\Program Files (x86)\OpenCPN\plugins\br24radar_pi.dll
8:13:10 AM:   C:\Program Files (x86)\OpenCPN\plugins\br24radar_pi.dll
              API Version detected: 114
              PlugIn Version detected: 1208
8:13:10 AM: BR24radar_pi: Target trail value 0 = 0 revolutions
8:13:10 AM: BR24radar_pi: Target trail value 0 = 0 revolutions
8:13:10 AM: BR24radar_pi: Target trail value 0 = 0 revolutions
8:13:10 AM: BR24radar_pi: Target trail value 0 = 0 revolutions
[... 76 lines cut ...]
8:13:14 AM: BR24radar_pi: main timer
8:13:14 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: SetPositionFixEx var=-14.000000 var_wd=1
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:15 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: Radar reports status STANDBY
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: SetPositionFixEx var=-14.000000 var_wd=1
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: main timer
8:13:16 AM: BR24radar_pi: main timer

NAHANNIV, 2017-09-10:
Could there be something about my hardware ?

http://minix.com.hk/en/products/neo-z83-4

It has Intell Graphics; I know OpenCPN has several exceptions for OpenGL when running on Intell Graphics ?

douwefokkema, 2017-09-10:
I am testing ow on a Intel I7 with simulated radar, so I can not zoom in/out myself, but the simulation contains many zooms. I will dust off an older Atom based laptop and try on that with the real radar, mid of the week I expect to be on board. And yes, there are issues with Windows Creator version and Intel graphics with OpenGL.

NAHANNIV, 2017-09-10:
I don't have creator version yet. I'm at 1607.

NAHANNIV, 2017-09-10:
I also have heading by Serial (USB) and quite a few AIS targets.

Here is a sample of my Serial data:

$AIMWV,343.2,R,11.1,N,A*02

$APHDM,121.3,M*32

$AIVHW,,T,121.3,M,0.0,N,,K*5C

$AIVLW,4877.78,N,4877.78,N,,N,,N*45

$AIMWV,343.2,R,11.1,N,A*02

$GPRMC,123941.00,A,1200.16239,N,06144.09460,W,0.065,357.60,100917,14.81,W,D*0A

$GPVTG,357.60,T,12.42,M,0.065,N,0.120,K,D*14

!AIVDM,1,1,,A,B5NbRg000>qEq>QeoA>gcwl5oP06,0*0A

$GPGGA,123941.00,1200.16239,N,06144.09460,W,2,12,0.79,1.4,M,-38.7,M,,0000*68

$APHDM,121.5,M*34

$GPGSA,A,3,46,01,30,06,51,07,08,17,28,22,09,23,1.52,0.79,1.29*09

$AIMWV,343.2,R,11.1,N,A*02

$GPGSV,5,1,17,01,55,019,52,03,41,131,,06,11,223,25,07,59,218,39*78

$GPGSV,5,2,17,08,13,053,39,09,11,191,24,11,39,020,,16,02,124,*72

$GPGSV,5,3,17,17,21,294,40,19,07,273,,22,39,097,46,23,07,162,33*7B

$AIHDG,121.5,,,14.0,W*0A

$GPGSV,5,4,17,28,29,338,43,30,52,283,49,33,35,101,,46,47,258,42*7E

$GPGSV,5,5,17,51,36,258,40*45

$GPGLL,1200.16239,N,06144.09460,W,123941.00,A,D*72

$APHDM,121.6,M*37

$AIMWV,343.1,R,11.1,N,A*01

$GPGST,123941.20,1241,,,,0.48,0.38,1.2*5B

$APHDM,121.8,M*39

$AIMWV,343.1,R,11.1,N,A*01

$APHDM,121.9,M*38

$AIMWV,342.9,R,11.1,N,A*08

$APHDM,122.1,M*33

$AIMWV,342.8,R,11.1,N,A*09

$APHDM,122.2,M*30

$AIMWV,342.7,R,11.1,N,A*06

$APHDM,122.3,M*31

$AIMWV,342.5,R,11.1,N,A*04

$AIHDG,122.0,,,14.0,W*0C

[... 582 lines cut ...]
$AIVLW,4877.78,N,4877.78,N,,N,,N*45

$AIHDG,132.0,,,14.0,W*0D

$AIMWV,332.1,R,12.1,N,A*04

$GPRMC,123950.00,A,1200.16222,N,06144.09471,W,0.164,357.60,100917,14.81,W,D*00

$GPVTG,357.60,T,12.42,M,0.164,N,0.304,K,D*10

$GPGGA,123950.00,1200.16222,N,06144.09471,W,2,12,0.71,1.6,M,-38.7,M,,0000*68

$APHDM,132.2,M*31

$GPGSA,A,3,46,01,30,06,51,07,08,17,28,11,22,09,1.42,0.71,1.23*0B

$AIMWV,332.1,R,12.1,N,A*04

$GPGSV,5,1,17,01,55,019,51,03,41,131,39,06,11,223,34,07,59,218,47*78

$GPGSV,5,2,17,08,13,053,33,09,11,191,25,11,38,020,43,16,02,124,*7F

$GPGSV,5,3,17,17,21,294,40,19,07,273,,22,39,096,39,23,07,162,30*71

$GPGSV,5,4,17,28,29,338,48,30,52,283,48,33,35,101,,46,47,258,38*79

canboat, 2017-09-10:
John, can you please upload files instead of copy/pasting long strings in took keep the issue readable?

canboat, 2017-09-10:
> It has Intell Graphics; I know OpenCPN has several exceptions for OpenGL when running on Intell Graphics ?

That was mostly related to the '500' hardware in the original Atom, which was utter crap. The Intel HD is not that bad actually.

NAHANNIV, 2017-09-10:
I'm in a remote anchorage at the limit of WiFi connectivity.
I have tried to upload files, but usually cutting and pasting text into the comment has a better chance of posting.
Sorry,
JM.

douwefokkema, 2017-09-10:
Just another try. Here: https://www.dropbox.com/s/bswvb062vehklj6/br24radar_pi-2.3-5c778bb-win32.exe?dl=0 is the version from June this year which I used all summer. It does not have the ARPA and AIS targets on the radar windows yet. Interesting to know if problem is there as well.

canboat, 2017-09-10:
Also, AIS target count for you is relatively low. My AIS target count is 210, and this does not cause any issues, other than looking ugly:

[image]

canboat, 2017-09-10:
ok helgrind has found some unprotected cross-thread access, preparing a fix quickly.

canboat, 2017-09-10:
And here is the location for the latest build, with various small issues regarding memory and threading fixed.

https://www.dropbox.com/s/kzslq4fijwa68bl/br24radar_pi-3.910-win32.exe?dl=0

NAHANNIV, 2017-09-10:
The old version did not crash.

canboat, 2017-09-10:
And new 3.910 version? If it does crash, is this still related to the radar being on, or does it also crash if you turn the radar off, shutdown/start O, turn radar on? I can't see any code path that would make it different when the radar is already on, but we've been proven wrong for days ;-)

canboat, 2017-09-10:
I've just updated the 3.910 version with one that can hide the AIS and MARPA targets on the PPI window. Basically that is the biggest difference between Douwe's June version and the production release. 

Can you test with:
- Only one radar enabled (I think you already run like this)
- AIS/ARPA targets hidden (you can find this on the PPI menu View > Show/Hide AIS/ARPA)

NAHANNIV, 2017-09-11:
The new version is not crashing.

Some remaining questions:

1/ after copying back my original OpenCPN.ini file I was unable to get the RADAR A PPI window to display ? Tried deleting all the RADAR settings from the ini file, but then could not get the RADAR to work at all ? How did it get into this state ?

2/ Your picture shows AIS target names on the PPI window. Mine is not showing names?

douwefokkema, 2017-09-11:
Using your ini file I also had the radar A window hidden. It was outside the visible area. Go inside the ini file  and find Radar0WindowPosX= In my case this was -604, that is outside the visible area. Make it 100 or so. After running O, I found this position being reset to its previous value again, that was the value of the first floatx of the [AUI] section. Also correct this floatx to some decent value; 0 or 100, and all remained OK, radar A window visible again. Don't ask me why.

NAHANNIV, 2017-09-11:
OK, got the PPI window back. That happened sometime after updating to latest OpenCPN and plugin.

Should the plugin be displaying AIS target names on the PPI window as in the example posted by Canboat ? Or is that a test version ?

canboat, 2017-09-11:
If you have AIS labels on the chart they should be present in the PPI window as well. We’re calling the OpenCPN AIS drawing code, so it should be exactly the same — with all that entails. Unfortunately most black additions to the symbols are not very visible on the PPI window unless you change the background color in the preferences.

NAHANNIV, 2017-09-14:
Sorry If I closed this prematurely.
I was fairly certain it had been fixed.

I think a problem may have been introduced in one of the intermediate builds that I have been testing, but I have no way to go back to test this theory.

NAHANNIV, 2017-09-15:
@douwefokkema the crash only seems to happen if the scanner is still on when OpenCPN is re-started.

NAHANNIV, 2017-09-18:
@douwefokkema were you able to reproduce this crash ?

@canboat if I am the only one experiencing this crash, can we use the crash report mechanism to get you more details ? Or, is there anything else I can do to help debug it ?

douwefokkema, 2017-09-18:
Yes, I can reproduce now and then (on the boat), time consuming. But working on it right now.

canboat, 2017-09-19:
I tried to reproduce on Windows and macOS, but for me it works fine even with Douwe’s instructions.

I will look into distributing a debug build for Windows with a way to get the dump back to us. Maybe, just maybe, it will give us good information (if it is a memory issue it usually crops up later, and the stack dump only tells you there is an issue.)

douwefokkema, 2017-09-22:
I found a possible divide by zero, corrected it, but this was not the cause of the crash.
Then I found a possible array boundary violation when zooming and corrected it. But this was not the cause of the crash.
Then I found another array boundary violation, index had strange value, as if it was not initialized. But according to code and log it was initialized. But then I found that with running radar receive thread starts receiving and processing before all initialization is done on the main thread (an slower computer). So I put the initialization (ClearTrails) also in the start of the receive thread. Problem solved here.
Please try https://www.dropbox.com/s/wb50m26ock7n5em/br24radar_pi.dll?dl=0
But... I worked in the true trails section and can only test true trails it when the ship moves again, in a few days.

NAHANNIV, 2017-09-22:
@douwefokkema Looks Good on my computer so far, will continue to test.
Thanks.

douwefokkema, 2017-09-23:
The cause of the crash (early start of receive) that I indicated yesterday was wrong. Kees hinted me that the position might be invalid while shown as valid. This was indeed the case.
New version https://www.dropbox.com/s/25lb561uw82wci4/br24radar_pi.dll?dl=0
Sources same location as above.

NAHANNIV, 2017-09-23:
@douwefokkema tried the new version, have not been able to get it to crash. The last version did not crash either.

canboat, 2017-09-23:
Added some other fixes, this should be the final release if it does not break anything:

https://www.dropbox.com/s/jq9osfhcm5dim6k/br24radar_pi-3.923-win32.exe?dl=0

[1 more reply]

closed 2017-09-24 by canboat