keesverruijt/BR24radar_pi #285: AIS targets in RADAR window sometimes in wrong position

NAHANNIV, 2017-09-08:
This happened on my first trial of the new plugin. 
I looked at preferences, adjusted GPS offset.
When I tried the windows I noticed this.

Have not been able to reproduce it since.
Thought I should mention it, perhaps you have an Idea as to what would cause it ?

[image]

NAHANNIV, 2017-09-08:
Quote from CanBoat on OpenCPN forum:
"The picture shows the radar was in non stabilized head up mode. In that mode it should not draw AIS targets at all. Once you select a heading mode it should not allow you to choose this mode since you have a compass.

This is probably because you had it installed before and had head up mode selected, with compass this is now replaced with head up stabilized. We haven't tested upgrades enough, small bug."

NAHANNIV, 2017-09-08:
OK, the problem likely went away after I changed the orientation of the RADAR window.

canboat, 2017-09-08:
Pretty busy anchorage :-)

NAHANNIV, 2017-09-08:
Not too busy by local standards. 

In the screen capture I am using the MARPA function to keep track of boats anchored around me who do not have AIS. The wind is light and clocking around due to the passage of Hurricane IRMA far to the north; Perfect conditions for boats to go bump in the night.

Thanks again for all the effort you have put into this excellent plugin. 

Cheers,
JM.

commit 0d48b0cd (2017-09-08): Fix #285. Convert to HEAD UP STABILIZED mode on boot
when old mode was HEAD UP. Allow HEAD UP mode if DeveloperMode=1 in config file.
closed 2017-09-08 by canboat