keesverruijt/BR24radar_pi #286: Enhancement: Tie RADAR control to PPI Window.

NAHANNIV, 2017-09-11:
Can the RADAR control be tied to the PPI window if displayed ?

Or can the RADAR control windows be made dock-able in the main chart window ?

canboat, 2017-09-11:
> Can the RADAR control be tied to the PPI window if displayed ?

You mean if you move one, move the other? It works this way on Mac. It is (too) hard on Windows.

> Or can the RADAR control windows be made dock-able in the main chart window ?

Tried that, did not work well because of various issues.

What I recommend is to set 'Auto Hide' for the menu, this can be found in preferences, and place the menu on top of the menu button.

Long term there is a plan to change the current menu from something drawn by the OS (which means simple coding, just say 'wxButton' and presto) to something drawn by ourselves. This means we have more code, but also more flexibility to add things like vertical sliders. It is about one man-month work... and then we can easily make it part of a dockable separate canvas or ap part of the current PPI canvas.

NAHANNIV, 2017-09-12:
"more flexibility to add things like vertical sliders"
Eagerly awaited !

I was trying to think of what to suggest to make the current control more user friendly, but I'll just wait and see what you come up with.