keesverruijt/BR24radar_pi #288: CPU overload detection is broken

canboat, 2017-09-13:
We're seeing that the system to automatically slow down the # of refreshes seems broken. This can even cause lockups on slow systems such as Intel Atom.

canboat, 2017-09-13:
I have not tested on board yet and I don't have a dual radar PCAP, but the current 'master' looks a lot better in terms of performance. Before this last commit it could push way too many 'refresh' messages in the GUI queue, causing overload showing as 'frozen' app and unresponsiveness.

The new algorithm may need a bit more tweaking. Douwe is going to test with a 5 year old Atom N270 + Intel 500, and with his onboard computer this weekend. So we may need a few more tweaks once those results are in but the new code is more readable and logical, so this should not be an issue.

The current code can be tested with: 

https://www.dropbox.com/s/l5eglhyi8x98hkg/br24radar_pi-3.913-win32.exe?dl=0

NAHANNIV, 2017-09-13:
That seems to fix the problem.
Overlay A and PPI A not locking up.
Overlay B seems to work at refresh rate 5
Both PPIs and overlay <50% CPU.

Even on my wimpy little Atom :)

canboat, 2017-09-14:
OK great that it works for you. I am going to wait for Douwe to report back this weekend before I close this issue.

douwefokkema, 2017-09-14:
I tested at home with simulated radar (TCPReplay) on an Samsung  with Intel Atom N450 2GB, 32 bit Windows 10. A really slow system to modern standards.

First I tested with OCPN 4.5.0 and the June version of the pi (5c778). Later version of pi not compatible with this version of O.
Result: Very workable system, with refresh rate up to 4 and one radar window open.  OCPN showing fps of 7 to 10. Would be good enough as navigation system on board.

Next test With OCPN 4.8.0 and last master of pi. 
Result: There is a radar image, refreshed now and then with refresh rate set to 1. But slow system, panning of chart possible but fails often, relocation of the radar window slow. OCPN shows 1 fps. 
Next test: results in non responding OCPN, with half drawn radar window.

Test of OCPN 4.8.0 with radar_pi disabled. Very slow in panning the chart and zooming. Much worse compared with 4.5.0.

My conclusion: the problem of the unresponsiveness on smaller systems is in OpenCPN itself, not in the radar_pi.

douwefokkema, 2017-09-14:
But on faster systems the BR24radar_pi is better than ever before. No bugs found here yet. Tomorrow test with real radar on board.

canboat, 2017-09-18:
Douwe reported by phone that it works well on board as well, so closing this issue.

closed 2017-09-18 by canboat