keesverruijt/BR24radar_pi #298: Error on making release using VS2017

sha3sha3, 2018-06-26:
### Error on making release using VS2017
Hello guys,
I compiled the project using CMAKE on windows 10 64 bit and it success .
but after opening the solution with visual studio and try to make release it gives me about 7 errors in the code.
**I have not edited it**
i need to  compile the plugin and make a distribution. any help please?
ps: i am newbie in programming.
Thanks in advanced.

Hakansv, 2018-06-26:
@sha3sha3 
Hi. Well I use to build it using VS2017 no problem for resent code in master branch this repository.
My method use to be:
clone and build OpenCPN 4.99.0 e.g. resent master code for release.
Clone radar_pi and made \build.
(If already present delete what's in \build\*.*)
copy opencpn.lib from opencpn\build\release\ to radar_pi\build\
Build Radar_pi by cmake -G "Visual Studio 15 2017" -T v141_xp ..   
Open radar_pi\build\radar_pi.sln in VS2017.
Build for Release and then project PACKAGE (Right click "Project Only"->"Build only PACKAGE")
If you just want to test a build binary check Releases on this site. Note: OpenCPN 4.99.0 is required.
Good luck
Håkan

keesverruijt, 2018-06-26:
Also, this is the 'maintenance' mode of the code, we are no longer doing any updates to this codebase. The new code is in https://github.com/opencpn-radar-pi/radar_pi/

What do you mean with "make a distribution"?

Hakansv, 2018-06-26:
Oops. Thanks Kees. I was a bit quick here. Thought we where on radar_pi and not BR24radar_pi. (Nearly forgot this one :-)
And also is my wiki link above not correct. I'll pass the new wiki for compiling wxWidgets 3.1 and OpenCPN 4.99. hang on...

Hakansv, 2018-06-26:
I've now checked what's in the revised compiling OpenCPN Windows wiki as of my above link. 
The first part is now nicely updated to reflect a 4.99.0 build using wxWidget 3.1. So please follow all Pavels instructions. 
And, as said, I use to build using VS2017. (Release -- > Package) I assume the Package is what you mean with "make a distribution".
Håkan

sha3sha3, 2018-06-26:
@Hakansv @keesverruijt Thank you a lot 
yes i mean package i am sorry
I will follow your instructions
and again Thank you.

keesverruijt, 2018-06-26:
OK, let's close this here and take it to the new plugin's issue list, if you have any remaining problems.

closed 2018-06-26 by keesverruijt