keesverruijt/BR24radar_pi #299: What exactly are "spoke meters"?

msacarny, 2019-05-15:
When digesting spoke packets, range in "spoke meters" is converted to 'actual' meters. What exactly are spoke meters, and why was range expressed in them as opposed to standard meters?

douwefokkema, 2019-05-15:
From where do you get the term "spoke meters"? Unknown to me. In the spokes as received from the Navico radar, the range is encoded in decimeters, which we convert to meters.

msacarny, 2019-05-17:
Thank you. Let me clarify. For example, we command range = 500 m.

As I see it, 4G packet headers are interpreted as follows: 
In br24Receive::ProcessFrame(), line 240:
    large_range is determined to be 0x80
    raw_range = small_range = 0x0ca0 = 3232
    range_meters = range_raw / 4 = 808
Line 287:
    m_ri->ProcessRadarSpoke(..range_meters..)
Line 483:
    m_range.Update(convertSpokeMetersToRange(range_meters))
   Here, 808, apparently in 'spoke meters'  is converted back to 500 m by table lookup.

Why does the radar return such strange units for range in the packet header? What is the relationship between 'spoke meters' and 'actual' meters?

Hakansv, 2019-05-17:
"Why does the radar return such strange units for range in the packet header?"
Why??  Ask Navico. It is as it is. Here we just try to interpret. Why do you ask?
BTW. BR24radar_pi is so to say out. At least for OCPN v5.0 and up. Try https://github.com/opencpn-radar-pi/radar_pi

douwefokkema, 2019-05-19:
msacarny,
The BR24radar_pi code base is old and no longer maintained. We have not been using this code base for over a year now. If you have issues, please refer to the current radar_pi code.
Douwe

closed 2019-05-19 by douwefokkema